zhh2005757/FAST-LIO-Multi-Sensor-Fusion #3: Wheel speed information cannot be fused

youngsterkim, 2024-06-21:
Hello, may I ask that in the h_share_model function in your laserMapping.cpp file, the ekfom_data.h_x has been redefined many times, will this cause the wheel speed information to be unable to be fused with the lidar information? Is there a mistake here?

zhh2005757, 2024-07-01:
Thanks for your focusing. I set different cases for each ekfom_data.h_x definitions like "opt_with_wheel" or "opt_with_gnss", and each of these cases will end the function after finishing their updates. You can see "return" at the end of each cases. By the way, the gnss and wheel information updates are executed in the UndistortPcl function in IMU_Processing.hpp.

closed 2024-09-17 by zhh2005757

chenxiaocongAI, 2025-08-21:
zhao博，你好。请教两个问题。
1. h_share_model有三个更新函数，每次更新是如何知道更新哪个观测的。例如我更新雷达观测，但此时会进去opt_with_gnss，但该函数里面没有判断是否有数据进来。
2. V3D res = wheel_v_vec * s.wheel_s - s.offset_R_W_I.toRotationMatrix().transpose() * (s.rot.toRotationMatrix().transpose() * s.vel + angv_crossmat * s.offset_T_W_I);
最后的s.offset_T_W_I是不是需要把它转换到Imu坐标系下，再跟角速度相乘呢？

zhh2005757, 2025-08-27:
> zhao博，你好。请教两个问题。
>
> 1. h_share_model有三个更新函数，每次更新是如何知道更新哪个观测的。例如我更新雷达观测，但此时会进去opt_with_gnss，但该函数里面没有判断是否有数据进来。
> 2. V3D res = wheel_v_vec * s.wheel_s - s.offset_R_W_I.toRotationMatrix().transpose() * [...]

1. 我设置了不同的flag，例如opt_with_gnss，你可以找一下类似flag设置的地方，因为每个雅格比函数计算完就return了，所以能保证每次都是用一个函数。
2. s.offset_T_W_I就是定义在IMU坐标系下的轮速的外参平移

chenxiaocongAI, 2025-08-28:
谢谢 zhao博的讲解