zhh2005757/FAST-LIO-Multi-Sensor-Fusion #5: 关于轮速构建的残差对应的雅可比的推导

hongSS0919, 2024-07-02:
大佬，轮速融合中，轮速的观测构建的残差，对应各个状态的雅可比矩阵推导具体是怎么推导的呢？
可以参考哪些资料呢？我看您提到的论文，并没有很详细的关于轮速雅可比的推导
[image]

zhh2005757, 2024-07-02:
这部分网上有很多资料，主要一个就是姿态的李群求导，可以看SLAM十四讲或者其他文献，是基础数学知识。

hongSS0919, 2024-07-02:
> 这部分网上有很多资料，主要一个就是姿态的李群求导，可以看SLAM十四讲或者其他文献，是基础数学知识。

我尝试推了一下，在rot上，一致和您的结果差个负号。（0，3）位置的话好像是pos，（0，6）位置的才是rot吧？您这块是不是写错啦？

hongSS0919, 2024-07-02:
[image]
[image]

hongSS0919, 2024-07-02:
> 1. fastlio的状态变量是pos rot的顺序，和论文里不一致；
> 2. 代码里是右乘扰动，你这个是左乘？

1.状态量顺序问题是我看错了。抱歉
2.关于rot的求导您方便给我点指导么？我没推出来

hongSS0919, 2024-07-02:
[image]

zhh2005757, 2024-07-04:
其实左边1这部分推导是正确的，代码中是以残差的形式出现的，也就是等号左边减去等号右边构建的残差，这样就会差一个负号了

minghongss, 2024-07-04:
我已经推出来了，感谢大佬开源，为社区增加了更多的相关学习资料。我将关闭这个问题。

closed 2024-07-04 by minghongss

zhh2005757, 2025-09-01:
> > 1. fastlio的状态变量是pos rot的顺序，和论文里不一致；
> > 2. 代码里是右乘扰动，你这个是左乘？
> 
> 你好，请问在哪看到代码中状态变量是这个顺序，我在`common_lib.h`中看`StatesGroup`的定义都是R、P、V、bg、ba、g，谢谢

use-ikfom.hpp

Fafa-DL, 2025-09-01:
> > > 1. fastlio的状态变量是pos rot的顺序，和论文里不一致；
> > > 2. 代码里是右乘扰动，你这个是左乘？
> > 
> > 
> > 你好，请问在哪看到代码中状态变量是这个顺序，我在`common_lib.h`中看`StatesGroup`的定义都是R、P、V、bg、ba、g，谢谢
> 
> use-ikfom.hpp

谢谢，找到了，目前我想做零速检测（因为无轮速信息），当某时间内imu加速度和角速度均值符合阈值时调用零速更新，如果延用本框架轮速更新的处理逻辑，是否残差直接可定义为(0.0, 0.0, 0.0) - s.vel，且雅可比只需考虑对速度求导（-I）不用考虑旋转和陀螺仪零偏

zhh2005757, 2025-09-01:
> > > > 1. fastlio的状态变量是pos rot的顺序，和论文里不一致；
> > > > 2. 代码里是右乘扰动，你这个是左乘？
> > > 
> > > 
> > > 你好，请问在哪看到代码中状态变量是这个顺序，我在`common_lib.h`中看`StatesGroup`的定义都是R、P、V、bg、ba、g，谢谢
> > 
> > 
> > use-ikfom.hpp
> 
> 谢谢，找到了，目前我想做零速检测（因为无轮速信息），当某时间内imu加速度和角速度均值符合阈值时调用零速更新，如果延用本框架轮速更新的处理逻辑，是否残差直接可定义为(0.0, 0.0, 0.0) - s.vel，且雅可比只需考虑对速度求导（-I）不用考虑旋转和陀螺仪零偏

可以的