zhh2005757/FAST-LIO-Multi-Sensor-Fusion #18: 关于GPS融合打开后，估计的雷达与imu外参变化比较大的问题

jingyun107, 2025-06-12:
最近测试数据的时候发现，如果进行gps融合时候开启了雷达IMU的外参估计，本来确定的雷达与IMU之间的外参，尤其是平移量就会发生比较大的变化，偏移给定的初始值。关闭这个外参估计后，发现最后得到的点云地图会有比较大的畸变。想问一下这个是由于gps的时间戳没有同步导致的吗？（数据集里面的gps是rtk定位信息，定位也比较可靠）希望能指导一下

zhh2005757, 2025-06-17:
如果时间对齐的话，那有可能就是外参不准了。如果在lio模式下外参在线估计是正常的，而融合gnss后lio外参估计不正常，那可能是gnss和imu的杆臂不正确导致的，建议量一个准确的杆臂再测试下。其实lio外参在线估计不需要开启gnss就能估计好，因为lio的内符合度通常还是比较好的，尤其在短时间内。

jingyun107, 2025-06-18:
我测试发现我不加入gps约束，只是单纯跑fastlio2算法得到的雷达IMU的外参也是在变化，不知道是什么原因导致的，之前认为雷达IMU外参是不变的，您遇到过这个问题吗？
初始的外参为：0.04165, 0.02326, -0.0284;
运行之后的外参估计：
[image]

zhh2005757, 2025-06-18:
这种情况得确认一下是初始外参更准确还是估计后的外参更准确了，尤其是平移。如果是估计的不准确，可能的原因一个是初始的时候是否IMU存在扰动，或者不是静止开始运动，会导致刚开始产生错的估计，另外就是由于采用滤波的框架，fastlio的外参收敛比较慢，通常需要较大的运动激励才能收敛。