zhh2005757/FAST-LIO-Multi-Sensor-Fusion #20: Urban03 城市数据集跑的时候，会存在飘的情况

chenxiaocongAI, 2025-08-21:
大哥，您好。 我在跑你开源贡献的代码，发现没有轮子作为约束的时候，只用雷达，IMU，RTK，出现容易把图拉飘的情况
[image]

zhh2005757, 2025-08-27:
kaist数据集主要因为有太多动态物体存在，所以激光惯性里程计会漂移，加轮速可以抑制一部分激光的错过观测，但是本质上还是激光匹配出现了错误。你可以单纯跑fastlio看一下，然后调一下参数等等。

chenxiaocongAI, 2025-08-28:
谢谢大哥的回复。我试了单跑fastlio2是不会飘的，只有在不加轮速，而加入rtk会有这种情况，应该是协方差太大导致，我尝试把协方差条件阈值降低，可以减缓飘的现象。
1. 另外除了使用协方差去判断rtk的可用性，还有什么好的方法不。
2. 如果要把RTK加入到LIO里面，使用因子图建立相对约束，是否有力抵抗outler rtk（即使协方差较小）的影响。相比与这种直接融合的形式。
3. 另外代码里面把第一个rtk数据作为origin，如果第一个点是outlier，岂不会导致这个观测误差，直接影响到后面的估计
下面有个好奇怪的现象，就是有个飘起来的点云，不知道是为何导致。

[image]

zhh2005757, 2025-08-28:
1. 涉及到GNSS的故障识别和剔除了，有很多办法例如通过比较信息和状态标准差的sigma范围，采用卡方检验剔除等；
2. 是的，因子图利用窗口内多个观测再结合鲁棒核函数可以抵抗一部分RTK错误观测
3. 会的，现有的算法激光雷达是和局部地图匹配，因此需要保证初始的地图构建是准确的，否则建图就不会不一致；这个飘在空中的点云我不确定是传感器数据的问题还是点云拼接的问题，貌似我在其他算法对kaist数据的实验中也看到了这个现象。

zhh2005757, 2025-08-28:
其实和原点关系不大，我的意思是激光点云会被RTK的跳变拉偏，导致不一致的情况从而影响后续点云匹配，这个可能发生在初始点也可能发生在后续RTK的跳变时刻

closed 2025-12-18 by zhh2005757